manolofdez/AsyncBluetooth #6: Swift Playgrounds 4 on iPad cannot add AsyncBluetooth

bricklife, 2022-02-03:
I'd like to use your AsyncBluetooth with Swift Playgrounds 4 on iPad, but I cannot do that. It doesn't seem to be able to import any Swift Package released by tags with alphabetic characters, like `v1.1.0`, as follows:

[image]

I was able to import my cloned repository re-tagged with `1.1.0` as follows:

[image]

If there is no particular reason, could you use only numbers for tagging?

manolofdez, 2022-02-04:
Whoa, that's surprising! Version tags with a "v" prefix are very common. Let me do some research. I'll keep this issue open and post back here.

bricklife, 2022-02-04:
@manolofdez Thank you for your response!

Yeah, I think "v" prefix is very common too, but Apple says:

> To create a version tag, tag the last commit with the package version. A package version is a three period-separated integer. An example is 1.0.0.

https://developer.apple.com/documentation/xcode/publishing_a_swift_package_with_xcode

I found some of the same issues as follows:
https://github.com/AudioKit/AudioKit/issues/2643
https://github.com/maxxfrazer/FocusEntity/issues/31

manolofdez, 2022-02-06:
Ok, I made a new release without the "v" prefix and will be using this naming style for releases going forward. Thanks for letting me know about this!

closed 2022-02-06 by manolofdez

bricklife, 2022-02-07:
Thank you very much! I just confirmed that my iPad can import your library 🚀