manolofdez/AsyncBluetooth #11: How do I read/write to a peripheral?

soltesza, 2022-08-08:
As far as I can tell, there are no examples for reading from and writing to a peripheral, and no documentation whatsoever for the writeValue method. I'm having trouble receiving data from my peripheral but I'm not sure if I'm even using the readValue method correctly.

manolofdez, 2022-08-09:
Hi! There are some examples in the [README.md](https://github.com/manolofdez/AsyncBluetooth/blob/main/README.md) from which you can see how to use the `readValue` method. The `writeValue` works very similarly to readValue. You can find more in depth examples in the [AsyncBluetooth Cookbok](https://github.com/manolofdez/AsyncBluetoothCookbook).
Let me know if after reading these you're still running into issues.

soltesza, 2022-08-09:
Hi thanks for the response. Can you point me to where in the cookbook it covers reading and writing? Thanks for your help

closed 2023-01-12 by manolofdez