manolofdez/AsyncBluetooth #33: Crash while writing on service to which we have subscribed notify on update value

grhashmi5, 2023-09-04:
Hello,
I have subscribed to a characteristic to get a notification when its value is updated.
However, When I try to write to that characteristic, I get crash in the library.
Here is a screenshot of that crash.
[image]

manolofdez, 2023-09-04:
Hi! I unfortunately haven't been able to reproduce this. Do you mind posting the step necessary to reproduce this crash?

manolofdez, 2023-11-04:
I suspect this is related to [this](https://github.com/manolofdez/AsyncBluetooth/issues/37) issue. Hopefully the same fix applies to this too.

closed 2023-11-04 by manolofdez