manolofdez/AsyncBluetooth #35: Crash when switching off bluetooth while scan is ongoing

dev-pac, 2023-10-20:
Version: from 1.7.0 (not tested earlier)

Steps to reproduce:
- Launch a scan
- on your device go to the settings to switch off bluetooth

result:

Thread 5: Fatal error: SWIFT TASK CONTINUATION MISUSE: scanForPeripherals(withServices:options:) tried to resume its continuation more than once, throwing operationCancelled!

in CentralManaager().scanForPeripherals(...) in the catch block, line 87

expected:
- scan is "paused"

dev-pac, 2023-10-20:
Modification: working correctly in 1.4.1

manolofdez, 2023-10-21:
Aha! I know why this is happening!! `scanForPeripherals` will resume with an async stream, but the `enqueue` call stays awaiting until the stream completes. Turning off Bluetooth does not terminate the stream. So you're probably calling `cancelAllOperations` when the bluetooth is turned off, which tries to send an `operationCancelled` to the enqueued operations. But since the `scanForPeripherals` continuation has already been used, the app crashes.

I'm going to want to re-think the strategy here, but I have a quick workaround for this. Stay tuned....

closed 2023-10-21 by manolofdez

dev-pac, 2023-10-23:
Cool; thanks