manolofdez/AsyncBluetooth #45: Feature request: Changing behaviour of the logger

VincentBeltman, 2024-04-05:
Hi! We'd like to get more control over the logging of this library. Currently it posts a lot of logging in our console, that we'd like to more easily filter out. Furthermore, we'd like to remove the logging altogether in release builds for various reasons. Simply being able to disable logging through a boolean would be enough for our use cases, but I guess that giving the developer the ability to overwrite the logger could also work.

What's your opinion about this? Is this something you can add?

manolofdez, 2024-04-11:
Yes! I thought about this early on and completely forgot about it. I agree this library can be quite noisy, and it would be ideal to provide more control over the logs. I'll cook something up and update here. Stay tuned...

manolofdez, 2024-04-13:
I need to think a bit more about filtering, but for now this should help with the noisiness.

closed 2024-04-13 by manolofdez