manolofdez/AsyncBluetooth #58: peripheral.setNotifyValue(Bool, forCharacteristicWithCBUUID:) method never responds

Dan1973KS, 2024-08-13:
Hi Manolo!

Thanks for silencing the compiler set to "strict concurrency check enabled" with the latest update!

Now as I delve deeper into the topic, I wanna write to characteristics. 
The above method looks really concise and convenient to achieve this goal.
Before calling it, I ensured that the peripheral and the aimed characteristic are really present by discovering them manually.
Although your method will do it anyway if necessary...
But this function never returns - throws no error either.

Please help - thanks in advance!

manolofdez, 2024-08-14:
Hi! The `peripheral.setNotifyValue` will enable/disable notifications for a characteristic, so whenever it changes you get the value through the `characteristicValueUpdatedPublisher`. If you want to write to a characteristic, look into `Peripheral.writeValue`:

```
try await peripheral.writeValue(
    value,
    forCharacteristicWithUUID: UUID(uuidString: "")!,
    ofServiceWithUUID: UUID(uuidString: "")!
)
```

Dan1973KS, 2024-08-14:
Thanks for your quick response. What you described above is exactly what I did. You mentioned it already in the "readme" file...
Here is my code:

await peripheral.characteristicValueUpdatedPublisher
                .filter { $0.cbCharacteristic.uuid == self.commandResponseUUID }
                .map { try? $0.parsedValue() as Data? }
                .sink { value in
                    print("Value updated to \(String(describing: value))")
                }
                .store(in: &cancellables)
            
try await peripheral.setNotifyValue(true, forCharacteristicWithCBUUID: commandResponseUUID, ofServiceWithCBUUID: serviceUUID)

print("Set notify to true")

When executing it never reaches the last print statement.
Maybe you're missing the delegate method peripheral(_:didUpdateNotificationStateFor:error:) in your framework? 
In case something went wrong, no error will be thrown, right...!?

Dan1973KS, 2024-08-14:
Sorry, Manolo!
I found out that I simply missed the do-catch-block in order to receive the appropriate error.
And I found the "missing" delegate method as well in your file!
Thank you so much for your patience...

closed 2024-08-23 by manolofdez