manolofdez/AsyncBluetooth #59: Xcode 16 crash on scan

dcilia, 2024-08-22:
Hi, I am getting a crash in Xcode 16.1 beta (latest build) when calling ```scanForPeripherals...``` the function does not crash when called on Xcode 15.

manolofdez, 2024-08-23:
Hi, I just tested on Xcode 16.1 beta (16B5001e) and didn't get a crash. Do you mind sharing how your `CentralManager` is constructed and how you're calling `scanForPeripherals`?

dcilia, 2024-08-23:
Hi, I'm using ```2.0.0``` of the package.  I made a quick project to reproduce.  Attaching it here.
[AsyncBT.zip](https://github.com/user-attachments/files/16727978/AsyncBT.zip)

```
struct ContentView: View {
    var body: some View {
        VStack {
            Image(systemName: "globe")
                .imageScale(.large)
                .foregroundStyle(.tint)
            Text("Hello, world!")
        }
        .padding()
        .task {
           let stream = try? await CentralManager().scanForPeripherals(withServices: nil)

        }
    }
}
```

dcilia, 2024-08-23:
Sorry forgot to mention I can reproduce using Mac Catalyst destination

manolofdez, 2024-08-23:
Thanks for posting the sample project. I was able to reproduce the crash with that project, though like you said, only when using the Mac Catalyst destination. That said, it seems like it's not a bug with `AsyncBluetooth` specifically. It's also reproducible if you replace the task modifier with:

```
.task {
    try? await withCheckedThrowingContinuation { continuation in
        continuation.resume()
    }
}
```

I'll keep an eye out, but I suspect there's nothing we can do here.

dcilia, 2024-09-22:
It's been resolved by Apple for a future Xcode release.

closed 2024-09-22 by dcilia