manolofdez/AsyncBluetooth #62: Possible concurrency Issue when receiving a stream of values from characteristic from version 3.0.0

andr3a88, 2024-10-17:
Hi, 

i've updated AsyncBluetooth from version `1.12.0` to `3.0.0` and i'm noticing a problem when receiving data from a characteristic. 

The data received is not consistent, some packages are double and some packages are missing.

After debugging i figure out that the problem may be is inside the method `func peripheral(_ cbPeripheral: CBPeripheral, didUpdateValueFor characteristic: CBCharacteristic, error: Error?)` on `PeriphepalDelegate.swift`, probably a concurrency issue.

It seems that the characteristic value sent to the client do not correspond to the characteristic value received from `didUpdateValueFor` of `CBPeripheralDelegate`.

As you can see from the logs below, for some reason the value of the characteristic inside the `Task` is an old value received previously and not the one from `characteristic` parameter of the delegate. The Task does not capture the correct value.

Logs:
````
##### value from didUpdateValueFor 88001A00811E116718E22802FBB13930324D ✅
##### value ---> task 88001A00811E116718E22802FBB13930324D ✅
##### value from didUpdateValueFor 88001B00831E116719F62813030000000000 ❌missing
##### value from didUpdateValueFor 88001C006B1F11671832300234CDB35E5B7C ✅
##### value ---> task 88001C006B1F11671832300234CDB35E5B7C  ✅
##### value ---> task 88001C006B1F11671832300234CDB35E5B7C ❌double
##### value from didUpdateValueFor 88001D006C1F1167193D3013030000000000  ✅
##### value ---> task 88001D006C1F1167193D3013030000000000 ✅
##### value from didUpdateValueFor 88001E00CF1F11671852330234CDB35E5B7C  ❌missing
##### value from didUpdateValueFor 88001F00D01F1161111D3313030000000000 ✅
##### value ---> task 88001F00D01F1161111D3313030000000000 ✅
##### value ---> task 88001F00D01F1161111D3313030000000000 ❌double
````

The function from `PeripheralDelegate.swift` with the `print` statements used to debug:

````
    func peripheral(_ cbPeripheral: CBPeripheral, didUpdateValueFor characteristic: CBCharacteristic, error: Error?) {
        print("##### value from didUpdateValueFor \(characteristic.value!.bytes.toHexString())")
        Task {
            print("##### value ---> task \(characteristic.value!.bytes.toHexString())")
            do {
                if characteristic.isNotifying {
                   await self.context.characteristicValueUpdatedSubject.send(Characteristic(characteristic))
                }
                let result = CallbackUtils.result(for: (), error: error)
                try await self.context.readCharacteristicValueExecutor.setWorkCompletedForKey(
                    characteristic.uuid, result: result
                )
            } catch {
                guard !characteristic.isNotifying else { return }
                Self.logger.warning("Received UpdateValue result for characteristic without a continuation")
            }
        }
    }
````

Utils to print in hex format
````
extension Data {

    var bytes: [UInt8] {
        [UInt8](self)
    }
}

extension UInt8 {

    func toHexString(use0x: Bool = false) -> String {
        String(format: "%02X", self)
    }}

extension Array where Element == UInt8 {
    func toHexString(use0x: Bool = false) -> String {
        map { $0.toHexString(use0x: use0x) }.joined()
    }
}

````

Reverting to `1.12.0` I've no evidence of the problem.

Let me know how i can help!

andr3a88, 2024-10-17:
I think I found the issue, I made some changes to validate an idea for the fix. 

Let's take as an example a case where a BLE device sends many packets one after the other (unsolicited answers) to the central.

First of all, `characteristic` is a reference type that is passed around all the Tasks created, the tasks are not scheduled serially and can run with some delay due to OS resource scheduling. Meanwhile CoreBluetooth continue to push new data calling the delegate and changing `characteristic.value` inside the tasks, so this is why we may have some cases where Tasks sends the same `characteristic.value` to the client.

To overcome this I move the creation of the wrapper `Characteristic` outside the Task, but also in this case we are creating a Struct injecting the characteristic (`CBCharacteristic`) that is a reference type, so  also `safeCharacteristic.value` changes because is a computed property that compute the value of the CBCharacteristic.

Finally I tried to replace the computed `value` of `Characteristic` with  an instance property `public let value: Data?` and passing them via initializer, in this case the `value` is copied and don't change.

 From my logs I can see that it's working. 

Below the code:

`PeripheralDelegate.swift`
````swift
    func peripheral(_ cbPeripheral: CBPeripheral, didUpdateValueFor characteristic: CBCharacteristic, error: Error?) {
        print("##### outer \(characteristic.value!.bytes.toHexString())")
        let safeCharacteristic = Characteristic(characteristic, value: characteristic.value)
        Task { @MainActor in
            print("##### inner ---> \(safeCharacteristic.value!.bytes.toHexString())")
            if safeCharacteristic.isNotifying {
               await self.context.characteristicValueUpdatedSubject.send(safeCharacteristic)
            }

            do {
                let result = CallbackUtils.result(for: (), error: error)
                try await self.context.readCharacteristicValueExecutor.setWorkCompletedForKey(
                    safeCharacteristic.uuid, result: result
                )
            } catch {
                guard !safeCharacteristic.isNotifying else { return }
                Self.logger.warning("Received UpdateValue result for characteristic without a continuation")
            }
        }
    }
````

`Characteristic`
````swift
public struct Characteristic: Sendable {
    public let cbCharacteristic: CBCharacteristic

    public let value: Data?

    public init(_ cbCharacteristic: CBCharacteristic, value: Data?) {
        self.cbCharacteristic = cbCharacteristic
        self.value = value
    }
    
    /// The Bluetooth-specific UUID of the characteristic.
    public var uuid: CBUUID {
        self.cbCharacteristic.uuid
    }
    
    public var properties: CBCharacteristicProperties {
        self.cbCharacteristic.properties
    }
    
    /// The latest value read for this characteristic.
//    public var value: Data? {
//        self.cbCharacteristic.value
//    }

    /// A list of the descriptors discovered in this characteristic.
    public var descriptors: [Descriptor]? {
        self.cbCharacteristic.descriptors?.map { Descriptor($0) }
    }

    /// A Boolean value that indicates whether the characteristic is currently notifying a subscribed central
    /// of its value.
    public var isNotifying: Bool {
        self.cbCharacteristic.isNotifying
    }
}

````

manolofdez, 2024-10-18:
Hi! Great find! You're absolutely right. We had this issue before, [fixed it](https://github.com/manolofdez/AsyncBluetooth/pull/10), and I accidentally introduced it [again](https://github.com/manolofdez/AsyncBluetooth/pull/56) 🤦🏼‍♂️. Let me fix it...

manolofdez, 2024-10-18:
Let me know if this fixes it for you 👍🏼

closed 2024-10-18 by manolofdez

andr3a88-amk, 2024-10-18:
@manolofdez yes it is all solved! ✅

andr3a88, 2024-10-18:
@manolofdez yes it is all solved! ✅

tylerjames, 2024-11-22:
Hi, I think I'm encountering a similar problem on 4.0

I have a peripheral that returns a response in multiple parts that needs to assembled together after I've received the last part. It seems that very frequently I will miss one of the middle parts resulting in an invalid JSON once the parts are assembled.

I made a local copy of the library and added a print in the `didUpdateValueFor` delegate method in PeripheralDelegate and I added a print when my subscribed `characteristicValueUpdatedPublisher` receives a value. The low level delegate method receives messages that somehow never make it through the publisher. Always some messages that occur in the middle somewhere.

tylerjames, 2024-11-22:
Actually this was my own doing.

I was converting the publisher into any AsyncStream by adding:
```
.values
.eraseToStream()
```

But apparently `.values` does not buffer and if messages come in too fast some can get lost. My god this cost me two days trying to figure out what was going on. What have I done with my life?

Thanks for the library btw. It has made dealing with Bluetooth a lot less painful. Except when I introduce my own problems.