manolofdez/AsyncBluetooth #64: Is there a difference between readValue and subscribing to characteristUpdatePublisher when reading a single value from a characteristic?

tylerjames, 2024-11-27:
Sorry, this is more of a question and maybe a problem.

I have a scenario where I first write to a characteristic with a request. My firmware interprets the request and posts a response to the same characteristic. Some of our hardware is more memory constrained than others so sometimes the response will be a single message and sometimes it will be broken up into multiple messages.

When the peripheral posts multiple messages I can usually receive them all by subscribing to the `characteristicUpdatePublisher` but if it posts only a single message then the update publisher does not seem to receive it. 

To combat this I've had to add both a `readValue` after my `writeValue` to see if the whole response has been posted to the characteristic. 

Is this how things should be working and is this the right way to deal with it?

Thanks.

nhannam, 2024-11-30:
In a project I'm working on I'm using a similar pattern (although I call setNotify before subscribing, and never do explicit reads) and don't seem to be losing notifications from the peripheral.

There's a couple of things I think are worth mentionig:

- I have all the bluetooth interactions performed by an actor to ensure they are done one at a time, however I seemed to have problems with seeing unexpected responses to some requests.  I ended up putting this down to interleaving of requests as a result of actor reentrancy.  I've now placed all these requests behind an AsyncChannel and this seems to have cured my problems
- I still feel like there's a bit of a fudge in there because `characteristicValueUpdatedPublisher` is marked @MainActor whichreally isn't great when you are trying to do the work in the background.  I've had to mark AsyncBluetooth as preconcurrency.

tylerjames, 2024-12-18:
Is it possible that the issue is due to how Combine works? 

In my example above I've created the stream but I'm not actually subscribed to it until after I call `writeValue`?

My `await gatherMultipartResponse(fromStream: values)` contains a `for await value in values` line that I assume subscribes to the underlying publisher.

Is there maybe a better way to structure this sequence of calls so that I'm for sure subscribed to the publisher before the `writeValue` happens?

manolofdez, 2024-12-19:
Hi, 

@tylerjames yes, any value sent before subscribing to `characteristicValueUpdatedPublisher` will be lost. One thing you can do to get out of this is to use `async let` _before_  you write your value. This will continue execution until the value needs to be read. Something like this:

```swift
do {
  let values = await characteristicValueUpdatedPublisher
    .filter { $0.characteristic.uuid == MyService.commandCharacteristicId }
    .compactMap {
      $0.value
    }
    .compactMap { String(data: $0, encoding: .utf8) }
    .buffer(size: 10, prefetch: .keepFull, whenFull: .dropOldest)
    .values
    .eraseToStream()

  // It could just hang here if there is only one response
  async let response = gatherMultipartResponse(fromStream: values)
  
  try await setNotifyValue(
    true,
    forCharacteristicWithCBUUID: MyService.commandCharacteristicId,
    ofServiceWithCBUUID: MyService.serviceId
  )
  
  try await writeValue(
    request,
    forCharacteristicWithCBUUID: MyService.commandCharacteristicId,
    ofServiceWithCBUUID: MyService.serviceId
  )
  
  // Use the response
  return await response

  // ...
```

Let me know if that works!

tylerjames, 2025-02-20:
I _think_ this may have worked. It definitely works in the case where the messages sent from my device are broken up into multiple chunks. Unfortunately it's been hard for me to confirm that it works with a single message response because that really only happens on very old firmware and I haven't been able to replicate it recently.

Otherwise, it's certainly a cleaner solution and has worked with newer devices and newer firmware

closed 2025-02-20 by tylerjames