manolofdez/AsyncBluetooth #66: Not receiving response from discoverServices

andr3a88-amk, 2024-12-10:
Hi, I am experiencing an issue with my connection flow when using the library. Occasionally, the code gets stuck after calling `discoverServices`, and the process ends with a peripheral disconnection error (the peripheral has a timeout of 5 minutes)

Below is the relevant part of my code:
````
if peripheral.state != .connected {
    logger.debug("📶 Peripheral not connected. Connecting...", subsystem: Self.tag)
    try await centralManager.connect(peripheral, options: options)
}
logger.debug("📶 Connected to peripheral \(peripheral.cbPeripheral.description)", subsystem: Self.tag)

try await peripheral.discoverServices(sensorServiceIds.map { $0.cbUUID })
logger.debug("💡 Discover services for peripheral \(peripheral.identifier)", subsystem: Self.tag)
````

The logs indicate that `discoverServices` does not return the expected services. The connection seems successfully initiated (`state = connected`), but the discoverServices call appears to hang and does not return the services. This mainly occurs when `peripheral.state` is `connected`

````
17:14:09 📶 Connected to peripheral <CBPeripheral: 0x300d880d0, identifier = 24CF3D10-C283-018D-FB02-5BC4B0A024DE, name = Respiro, mtu = 247, state = connected>
17:19:42 didDisconnectPeripheral <CBPeripheral: 0x300d880d0, identifier = 24CF3D10-C283-018D-FB02-5BC4B0A024DE, name = Respiro, mtu = 23, state = connecting> | isReconnecting false | The connection has timed out unexpectedly
````

manolofdez, 2024-12-19:
Hi,
I am not aware of any issues with `discoverServices`. If you set a breakpoint in `PeripheralDelegate`'s line 34, does it get hit?

andr3a88-amk, 2025-01-10:
Hi @manolofdez, sorry for the delay. At the moment i can't figure out the issue. 

The only time when it happens is after a peripheral restore or after a consequent peripheral disconnection (caused by `The connection has timed out unexpectedly`) and reconnection. Then if I use the peripheral restored/reconnected sometimes i got `Service not found` and sometimes no responses from the peripheral.

nonameplum, 2025-01-13:
@andr3a88-amk could it be connected to this issue https://forums.developer.apple.com/forums/thread/742497 ?

andr3a88-amk, 2025-01-14:
@nonameplum yes is possible, but from my logs I cannot see a peripheral with same UUID but different memory address.

When a peripheral is restored i use the method `retrievePeripherals`, to get a fresh instance. but right after the restoration i have a disconnection: 

`let peripheralsRetrieved = centralManager.retrievePeripherals(withIdentifiers: [cbPeripheralRestored.identifier])`

````
08:37:38  Restored peripheral received: <CBPeripheral: 0x302fb6970, identifier = B19706B8-D5DE-BE36-560C-E909590D20E9, name = "", mtu = 247, state = connected>

08:37:39 didDisconnectPeripheral <CBPeripheral: 0x302fb6970, identifier = B19706B8-D5DE-BE36-560C-E909590D20E9, name = "" , mtu = 23, state = connecting> | isReconnecting false | The connection has timed out unexpectedly.
````

andr3a88-amk, 2025-01-22:
Hi @manolofdez, I have a question about the behavior of `connect` method.

Does this method return immediately after initiating the connection request (after the call), or does it stay in an await state until the connection is fully established?
```
try await centralManager.connect(peripheral, options: nil)
```

johnzorko, 2025-03-18:
I see what appears to be the same issue, where `try await peripheral.discoverServices(nil)` does not return. I see it when testing rapidly and unexpectedly going in and out of BLE range (via a Faraday bag). It seems that if a disconnect happens _just_ after the connection is established, but before service discovery is complete, this can happen. Here's the parts of my code where it shows up:

`        centralManager.eventPublisher
            .sink {
                switch $0 {
                case .didConnectPeripheral(let peripheral):
                    NSLog("** Connected to \(peripheral.identifier)")
                    
                    Task {
                        NSLog("** discovering services")
                        try await peripheral.discoverServices(nil).  // when the issue happens, it gets stuck here
                        NSLog("** discovered services")

                        for svc in peripheral.discoveredServices ?? [] {
                            NSLog("** discovering chars for service \(svc.uuid.uuidString)")
                            try await peripheral.discoverCharacteristics(nil, for: svc)
                        }
                        
                        if mgr.areRequiredServicesSupported(peripheral: peripheral) {
                            NSLog("** coolness - device is \(String(describing: peripheral.name)), \(String(describing: mgr.enso))")
                            mgr.isConnected = true
                        } else {
                            NSLog("** Required svcs not found for \(peripheral)")
                            try? await self.centralManager.cancelPeripheralConnection(peripheral)
                        }
`

I debugged into the PerjpheralDelegate and added some additional logging, and discovered that, when this happens, AsyncBluetooth's peripheral delegate's didDiscoverServices delegate is not called. This seems to indicate that either CBs discoverServices() API was never called, or the delegate was maybe removed before CB could call into it with the discovered services. I'm trying to debug further to see what's going on.

andr3a88-amk, 2025-03-20:
Hi, @johnzorko ahere are some cases where the **AsyncBluetooth** delegate no longer forwards events received from **CoreBluetooth**. 
In my use case, this happens after frequent connections and disconnections of the bBluetooth device. 

Since it is difficult to reproduce and I don’t have time to investigate, I had to fall back using CoreBluetooth directly, but there is definitely some kind of issue in `AsyncBluetooth` that causes the delegate to be lost.