manolofdez/AsyncBluetooth #79: writeValue sometimes does not return when sending multiple messages

tylerjames, 2025-04-10:
I've got a message that needs to be broken into chunks because it exceeds the MTU size.

To do this I'm looping over an array of values and calling a func that sends the message and awaits the response. It looks like this:
```swift 
func sendMessage<M: PeripheralDataConvertible, R: Decodable>(message: M) async throws -> R {
	try await withThrowingTaskGroup(
		of: R.self,
		body: { [weak self] group in
			guard let self else {
				throw BluetoothError.noConnectionToPeripheralExists
			}
			
			var output = ""
			group.addTask {
				do {
					let values = self.characteristicValueUpdatedPublisher
						.filter { $0.characteristic.uuid == SwidgetBLEService.commandCharacteristicId }
						.compactMap {
							$0.value
						}
						.compactMap { String(data: $0, encoding: .utf8) }
						.handleEvents(
							receiveSubscription: { _ in
								Logger.shared.verbose("💪💪💪 Subscribed to publisher!")
							},
							receiveOutput: { value in
								Logger.shared.verbose("📫📫📫 Publisher received value: \(value)")
								output.append(value)
							}
						)
						.buffer(size: 10, prefetch: .keepFull, whenFull: .dropOldest)
						.values
						.eraseToStream()
					
					async let asyncResponse: R = self.gatherMultipartResponse(fromStream: values)
					
					try await self.setNotifyValue(
						true,
						forCharacteristicWithCBUUID: SwidgetBLEService.commandCharacteristicId,
						ofServiceWithCBUUID: SwidgetBLEService.serviceId
					)
					
					Logger.shared.debug("✍️✍️✍️ Sending message")
					try await self.writeValue(
						message,
						forCharacteristicWithCBUUID: SwidgetBLEService.commandCharacteristicId,
						ofServiceWithCBUUID: SwidgetBLEService.serviceId
					)
					
					// MARK: Gather multipart response
					Logger.shared.debug("🌾🌾🌾 Gathering response from stream")
					let response = try await asyncResponse
					
					return response
					
				} catch {
					Logger.shared.debug("Send message error: \(error)", tags: [.ble])
					throw error
				}
			}
			
			group.addTask {
				try await Task.sleep(nanoseconds: 8 * NSEC_PER_SEC)
				Logger.shared.debug("❌ Send Message timed out!!")
				throw BLEMeshError.timeout
			}
			
			let result = try await group.next()
			
			Logger.shared.verbose("✅ One of the tasks completed!", tags: [.ble])
			group.cancelAll()
			
			if let result {
				return result
			} else {
				throw BLEMeshError.timeout
			}
		}
	)
}
```

And the loop that calls it looks like this:
```swift
for (index, certPart) in certParts.enumerated() {
	let response: ErrorCodeResponse = try await scanResult.peripheral.sendMessage(message: certPart)
	Logger.shared.debug("Cert part response: \(response)", tags: [.device, .ble])
	try await Task.sleep(milliseconds: 500)
}
```

This works fine for single messages but in the loop it will sometimes get stuck on the `writeValue` line after successfully sending several messages.

I have no idea why it's hanging there. Does it get stuck if it does not receive the response or are there messages just getting enqueued but not dequeued? It's very frustrating because sometimes it will make it all the way through but very often it just gets stuck there. I implemented a retry mechanism and it never was able to successfully send a message again once it got stuck

Any idea what might be causing this? Or is there a better way to send multiple messages?

Thanks

Unrelated:
Why does Github insist on making tabs into 8 spaces? Looks absurd

tylerjames, 2025-04-11:
I tried making a fairly barebones CoreBluetooth client as well that just uses continuations to make it async and it had the same problem. I don't know if it's my peripheral or if there's some queuing or checking that i need to be doing

bo2themax, 2025-08-17:
> I tried making a fairly barebones CoreBluetooth client as well that just uses continuations to make it async and it had the same problem. I don't know if it's my peripheral or if there's some queuing or checking that i need to be doing

If your message has a write type `.withoutResponse`, it probably has something to do with `peripheralIsReady(toSendWriteWithoutResponse:)`, since this library doesn't handle that, it may cause hanging after setting complete for writeCharacteristicValueExecutor. For a typical CoreBluetooth application, it will handle that in a kind of loop of resending the data that `failed` since CoreBluetooth has no error callback for that type of message.

I’m trying to implement a Peripheral Manager in the same manner, maybe able to fix this along the way. Will update later.